TeXitoi/keyberon #13: USB won't register anything beyond the first report

TyberiusPrime, 2019-08-03:
This isn't a bug in keyberon, but a call for help :).

I'm hacking a firmware based on keyberon and a library
called keytokey (unreleased...) that turns key-streams into usb key outputs,
and is supposed to abstract away from the keyboard firmware and allows
unit testing of arbitrary keyboard inputs. That part is mostly done, and should
support most of what QMK offers in this regard.

My issue is with the USB-Hid stuff - 
I'm getting the computer to read the first report, but as soon as I want to send another one (be it empty or another key press), my firmware hangs and the computers autorepeat keeps repeating that first keypress. 

Perhaps you might have run into a similar problem?

Code is at https://github.com/TyberiusPrime/stm32f103_k2k/tree/master/src
guess you,d want to look into usbout first.

Thank you so very much for getting HID working on Rust+Stm32!

TyberiusPrime, 2019-08-04:
I found it! For posterity, you can't write to the Hidclass in a loop - you need to push your reports into a buffer and send one each time USB_LP_CAN_RX0 comes back. Straightforward, actually, if you understand what's going on.

closed 2019-08-04 by TyberiusPrime

TeXitoi, 2019-08-04:
Great! I didn't had the time to look at your problem, sorry.

Please point to me your keytokey lib when you can.

TyberiusPrime, 2019-08-05:
Thanks for your interest, I've uploaded the library: https://crates.io/crates/keytokey
https://github.com/TyberiusPrime/KeyToKey

The basic's are in place and it will be my daily driver starting tomorrow,
but the more advanced functions are largely untested and their API will probably change.

I welcome all feedback - my Rust isn't strong yet.