TeXitoi/keyberon #19: Mouse function

TssT16, 2020-05-04:
First of all thanks for the code, I love my DIY keyboard with own keybindings,
my question is now is there a why so simulate the mouse.
So that I have mouse-like with arrow keys.
Or is that in plan?

TeXitoi, 2020-05-04:
Oh, great! Can I have a photo of tour keyboard to add it to the gallery? Did you also use the 3d printed case?

For the mouse, yes and no.

Basically, I want to provide a library and not a framework, so mouse functionality would be in a different crate.

But I want to provide user defined actions on the keyboard, for non keyboard features as rgb, reset, backlight or mouse.

I didn't find a great interface for it, but the idea would be to add a `Custom(T)` variant to the Action enum, allowing the used to create whatever he want. The difficult part of the api is to provide the custom key events in an ergonomic way.

TssT16, 2020-05-11:
Shure here is my first prototype built of a wood piece but for the moment I had no diodes.  
So some times I have ghosting I would say ones a week but for the moment it's ok.  
When can I test the first version of the mouse function?
By the way, I had uploaded [my keyboard code](https://github.com/TssT16/keyberon/blob/master/examples/keyberon48.rs) for the 4x12 keyboard. It is also a German layout.

[image]
[image]

gilescope, 2020-05-23:
Don't those bare wires stand a chance of shorting each other? I thought also some diodes were required ideally but maybe they're optional...

TeXitoi, 2020-05-23:
That's

TeXitoi, 2020-05-23:
That's enameled wire, so no risk of shorts.

TssT16, 2020-05-26:
@gilescope You can get this wires from a old Motor.

TeXitoi, 2020-05-26:
I personally use https://www.aliexpress.com/item/32960104613.html

TeXitoi, 2020-09-06:
So, to fix this issue, a Custom action must be possible, with an interface to expose event (action "pressed", action "released") and state (action is currently active).

riskable, 2020-10-26:
You can use my `SequenceEvent` enum for mouse events.  I'd be happy to start working on `MouseDown` `MouseUp` after my PR gets merged.  Technically it'd probably be something like

```rust
MouseDown {
    button: u8, // e.g. 1 for left click, 2 for middle, 3 for right, etc
}
MouseUp {
    button: u8,
}
MouseMove {
    x: u32,
    y: u32,
}
```

I'm pretty sure the mousewheel is just another button from a USB input event standpoint.  Just speculating at this point...  Might be a difference between something like *moving* the mouse and just having the mouse appear at a specific location.  It's been a while so I forget.

TeXitoi, 2020-10-26:
No, no direct mouse support in keyberon.

You'd need  #actions and not keycode to support mouse in

TeXitoi, 2020-11-08:
I think I have a plan for this feature:

Action with a Custom variant, that is a type parameter, default to core::convert::Infallible:

```rust
enum Action<T> {
    // ...
    Custom(T),
}
```

`Layout::tick()` will return a `CustomEvent<T>`

```rust
pub enum CustomEvent<T> {
    NoEvent,
    Press(T),
    Release(T),
}
```

Keyboard report will be extracted with the (already existing) `Layout::keycodes()`.

What do you think?

commit 855c8514 (2020-11-29): Custom Action

fix #19
PR #38 (2020-12-12): Custom Action
fix #19

TODO:
 - [x] ChangeLog
 - [x] test on real hardware
 - [x] improve doc
commit a7952125 (2020-12-12): Custom Action

fix #19
closed 2020-12-12 by TeXitoi