TeXitoi/keyberon #25: debouncer finetuning

gilescope, 2020-08-24:
Keen to hear others thoughts on debouncing. I have good days and bad days for debouncing. I think switching it from 5 to 7 helps.

But I'm also curious if there's other algorythms we could try. Typically it's always that I get two of the same letter. Happy to try some stuff, just not quite sure what else to try.

TeXitoi, 2020-08-25:
That's at 5 because mechanical switches are supposed to be stable after 5ms. A bit note can help if your switches are a  bit dirty or if your hamdwiring is a bit fragile.

The deboucer is as simple as possible. It should be enough for mechanical keyboard usage. Now, the debouncer is easily changeable as it is completely independent. You can easily play with it. Would be happy to ear about your experiments.

TeXitoi, 2020-08-29:
Mmm, maybe your double register is something else. Does it append only with hold-tap keys? It seems that, when releasing a normal key at about the same time as pushing a hold-tap key, and releasing this key before the tapping term, 2 tap keys are registered by the OS. Is it consistent with your problem? (I know that you use heavily the hold-tap feature)

I'll try to investigate on this, I have this issue, even with a debouce of 20, so it should not be a debouce problem as it only append on one key, a hold-tap one (I have only space and return as hold-tap, and it happen only on space, but space is much more frequent, and I don't hit enter very fast after another keys).

TeXitoi, 2020-08-30:
xev dump of my double space tap

```
KeyPress event, serial 28, synthetic NO, window 0x2c00001,
    root 0x17f, subw 0x0, time 275768344, (147,176), root:(828,208),
    state 0x0, keycode 41 (keysym 0x65, e), same_screen YES,
    XLookupString gives 1 bytes: (65) "e"
    XmbLookupString gives 1 bytes: (65) "e"
    XFilterEvent returns: False

KeyPress event, serial 28, synthetic NO, window 0x2c00001,
    root 0x17f, subw 0x0, time 275768544, (147,176), root:(828,208),
    state 0x0, keycode 65 (keysym 0x20, space), same_screen YES,
    XLookupString gives 1 bytes: (20) " "
    XmbLookupString gives 1 bytes: (20) " "
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x2c00001,
    root 0x17f, subw 0x0, time 275768552, (147,176), root:(828,208),
    state 0x0, keycode 65 (keysym 0x20, space), same_screen YES,
    XLookupString gives 1 bytes: (20) " "
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x2c00001,
    root 0x17f, subw 0x0, time 275768560, (147,176), root:(828,208),
    state 0x0, keycode 41 (keysym 0x65, e), same_screen YES,
    XLookupString gives 1 bytes: (65) "e"
    XFilterEvent returns: False

KeyPress event, serial 28, synthetic NO, window 0x2c00001,
    root 0x17f, subw 0x0, time 275768560, (147,176), root:(828,208),
    state 0x0, keycode 65 (keysym 0x20, space), same_screen YES,
    XLookupString gives 1 bytes: (20) " "
    XmbLookupString gives 1 bytes: (20) " "
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x2c00001,
    root 0x17f, subw 0x0, time 275768568, (147,176), root:(828,208),
    state 0x0, keycode 65 (keysym 0x20, space), same_screen YES,
    XLookupString gives 1 bytes: (20) " "
    XFilterEvent returns: False
```

Compressed version:

```
KeyPress event,   time 275768344, keycode 41 (keysym 0x65, e)
KeyPress event,   time 275768544, keycode 65 (keysym 0x20, space)
KeyRelease event, time 275768552, keycode 65 (keysym 0x20, space)
KeyRelease event, time 275768560, keycode 41 (keysym 0x65, e)
KeyPress event,   time 275768560, keycode 65 (keysym 0x20, space)
KeyRelease event, time 275768568, keycode 65 (keysym 0x20, space)
```

So, it seems that the sequence generated is:

```rust
[E]
// some times, after that, successive reports from the release of the hold-tap key
[E, Space]
[E]
[Space]
[]
```

I didn't find a way to reproduce that in a unit test.

TeXitoi, 2020-09-01:
I found my bug, it was a race condition on the layout object, but that's not relevant to the example case: I was doing a split keyboard and thus the usage was bad, but I didn't find any bug in layout.

TeXitoi, 2020-09-02:
For your double lettering problem, I'll check the solder joins first, as a bad join might cause such issues.

gilescope, 2020-09-03:
my soldering is indeed very suspect. I didn't realise that it could lead to double keys though. Taking it to 20 helps - I don't get doubles but I do get key misses. I've come to realise its the top left row so it has to be a diode junction somewhere. I get it now - the bad connection kicks in while key depressed so you get not one key but two

gilescope, 2020-09-03:
Two rounds of soldering onwards and it seems to be holding. Between the crators and molton plastic there's a few brave wires struggling to hold the line, and hold the line they do. At least for now, as they go into battle again at typeracer.com where we're determined to reclaim wps victories from yore and hopefully that maybe, just maybe, the lines will stand the test of time.

closed 2020-09-04 by gilescope