TeXitoi/keyberon #27: Implement nesting of layers

gilescope, 2020-10-02:
alt  -  f1

I have an alt key that also puts me in layer 2. if I then press a key to get to layer 3 as well then when I press the F1 key I get what's in layer 2 for that not what's in layer 3.

If instead I press layer 3 and then press alt (which in this case is a plain alt and not messing with layers) and then press the F1 key it all works fine. Would be great if both roads lead to the same place. Is this possible?

TeXitoi, 2020-10-02:
Can you point to tour layout? I don't understand the problem.

gilescope, 2020-10-06:
```&[
		&[	k(Escape),  k(Q),		k(C),   		k(H),    k(G),    	 	k(J),  	 	k(SColon), 	k(F),   	k(O),     	k(B),     	k(Quote),		k(Minus)  ],
		&[	l(1), 		k(S),		k(N),			k(R),    LSHIFT_T,   	k(P), 	   	k(W),   RSHIFT_U,   k(E),     	k(I),    	k(A),	k(Y) ],
		&[	k(LShift),  k(X),		k(M),			k(L),  	 k(D),    	 	k(Z),     	k(V),   k(K),   	k(Comma),   k(Dot),     k(Up),		l(3) ],
		&[  k(LCtrl),	k(LGui),	lm(2, &[LAlt]),	 k(Equal), k(BSpace), 	SPACE_L4,	k(Tab),	k(Enter),	k(Slash),  	k(Left), 	k(Down),	k(Right) ],		
	],
	&[
		&[k(Escape),	k(Kb1),		k(Kb2), k(Kb3), 			m(&[LShift,Kb7]),Trans, Trans,		m(&[LShift,Kb3]),   m(&[LShift,Kb6]), 				Trans,  			Trans,				k(Delete) ],
		&[k(Tab),		k(Kb4),   	k(Kb5), k(Kb6), 			m(&[LShift,Kb8]),m(&[LShift,NonUsBslash]), k(Grave), 	m(&[LShift,Kb6]),	m(&[LShift,Kb5]), 	m(&[LShift,Kb4]), 	m(&[LShift,Kb2]), 	m(&[LShift,Slash])],
		&[Trans,		k(Kb7),   	k(Kb8), k(Kb9), 			k(Kb0), 	Trans,		Trans, 		Trans,  			Trans,			 	Trans,			  	k(PgUp),		 	m(&[LShift,Kb1])],
		&[Trans,		Trans,		k(LAlt),Trans, Trans, 	m(&[LShift,Kb2]),  	k(Tab), 	k(Bslash), 			k(NonUsBslash),  	k(Home), 			k(PgDown),  		k(End)     ],
	],
		
	//alt layer
	&[
		&[  k(Escape),  k(Q), 		k(W),   	k(E),    k(R),    	k(T),   k(Y), 	 k(U),   	k(I),     	k(O),     	k(P), k(Delete)],
		&[  k(Tab),     k(A),		k(S),   	k(D),  	 LSHIFT_F,	k(G),   k(H),	RSHIFT_J,  L1_K,     	k(L),    	k(SColon),k(Quote)  ],
		&[	Trans, 		LSHIFT_Z,	k(X),		k(C),  	 k(V),    	k(B),   k(N),    k(M),   	k(Comma),   k(Up),    Trans,	l(3) ],
		&[	Trans,		Trans,		Trans,  	Trans,   Trans,  	Trans,  Trans,   	Trans,    Trans,   	Trans,   Trans,    	Trans ],
	],
	// function layer
	&[ 
		&[Trans,		 	k(F1),	k(F2), 	k(F3), 	 	k(F12), 	Trans,	Trans,					m(&[LShift,LBracket]),	m(&[LShift,RBracket]), Trans,  			Trans,				Trans],
		&[Trans, 			k(F4), 	k(F5), 	k(F6),   	k(F11),		Trans, 	m(&[LShift,NonUsHash]),	m(&[LShift,Kb9]),		m(&[LShift,Kb0]),  		k(NonUsHash),   m(&[LShift,SColon]),Trans],
		&[Trans,			k(F7), 	k(F8), 	k(F9), 		k(F10),  	Trans,	Trans,	 				k(LBracket),   			k(RBracket),  			Trans,			k(PgUp), 			Trans],
		&[Trans,			Trans,	k(LAlt),Trans,  	Trans, 		Trans,	Trans, 					Trans,					Trans,					k(Home), 		k(PgDown),  		k(End)    ],
	],

```

gilescope, 2020-10-06:
ah. I think I know why it doesn't make much sense. I am raising an issue that's using a feature that I've not PRed yet!
I added lm function that would change layer like l(2) and also press a key:
```
/// Layer and multiple keycodes
pub const fn lm(layer: usize, kcs: &'static [KeyCode]) -> Action {
    Action::LayerAndKeyCodes(layer, kcs)
}
 LayerAndKeyCodes(value, v) => {
                let _ = self.states.push(LayerModifier { value, coord });
                for &keycode in v {
                    let _ = self.states.push(NormalKey { coord, keycode });
                }
            }
```

So sorry about that - it has been useful (even though it only presses one key while changing layers rathat than the multiple keys that the code looks like it might.). So your code works flawlessly - it's just that my code when it invokes a layer and holds alt down at the same time and then I try and also hold my function layer key down to get to F1 is where it doesn't work. ( I don't expect you to fix this - but if you do spot something bad I am doing in these lines of code do tell)

gilescope, 2020-10-06:
Something that's not imediatly obvious is why have a whole layer with alt held down? The answer is purely for Alt-tab to work nicely!
(I do not have tab on the main layer in that position and for alt-tab to work alt needs to stay down while tab gets tapped a few times.)

TeXitoi, 2020-10-07:
I have implemented MultipleActions that you can use for your needed feature, and that is more generic.

TeXitoi, 2020-10-28:
Can we close this?

closed 2020-11-04 by TeXitoi