TeXitoi/keyberon #28: Chording aka two keys at the same time to create a keypress

gilescope, 2020-10-06:
Having realised on matrix keyboards that one can press two keys down fairly easily with one finger by aiming in between them I think that opens up opportunities that should be exlored as if you pick the rigth key combos the false positive rate would be rare.

I'm raising this issue as something that I'd like to see done - I'm not expecting anyone to else to do it (but feel free if the urge takes you). I might get around to doing it soon but am juggling a few things at the moment - not least of which is trying to learn 'hands down' as a keyboard layout.

TeXitoi, 2020-10-07:
I think a cording engine should just be inserted between debounce and layout, i.e. it takes events, and modify events. Adding a row to a layer for combo would allow to have new keys using combo, without touching anything else.

So you would say something like "0,0 and 0,1 at the same time will generate 7,0" and then you put what you want on LAYOUT[x][7][0].

gilescope, 2020-10-08:
That sounds like a great idea to keep complexity down.

I have this crazy idea to try: I want to see what happens when I make the third letter row of the keyboard be done when pressing the above letters in the corresponding column...

gilescope, 2020-10-08:
I tried implementing something right after the debouncer but no matter how close I think I press the keys I don't do it in the same cycle as the keyboard is scannig them so I think I need to have some period of <50ms where if they have very similar start times then one should try and combo them - I guess this is similar to the tap and hold functionallity so maybe layout.rs is the right place for it.
(Been having a little look over at https://github.com/qmk/qmk_firmware/pull/8591 and they seem to come to the same conclusions that I found)

TeXitoi, 2020-10-09:
You need to time a bit yes, but no need to go in layout. Just add a tick method to your proxy combo engine, and release the events after some time.

TeXitoi, 2020-10-09:
I understand this QMK PR as exactly that: combo before anything, no more at the same time as everything.

TeXitoi, 2020-10-09:
Even simpler: use the denounce to do the work for you: a denounce of 50ms will send you the evens grouped in 50ms exactly as you want.

gilescope, 2020-10-09:
Nice! That helps a lot, also removing a bug helped me no end. Now works almost flawlessly - the flaw is that if I press the combo but only release one key I get the combo rrrrrrrrrrrrrrrrrrrrrrepeating. I think I nead to monitor which combos are down as state and then lift them when any of the combo keys are released rather than waiting for all the combo keys to be released at the same time.

TeXitoi, 2020-10-09:
Great!

Seems legit to release the virtual key when the first real key is released.

gilescope, 2020-10-09:
nailed it :-) - now to give it a good old spin. 
(Will need to tidy it up before I PR it back).

gilescope, 2020-10-12:
Almost nailed it - there's a slight case that still occasionally locks up the keyboard. Will find it...

sevanteri, 2020-10-12:
> Nice! That helps a lot, also removing a bug helped me no end. Now works almost flawlessly - the flaw is that if I press the combo but only release one key I get the combo [...]

Hey, author of the QMK PR you linked to here!

I actually wanted to go the other route. The original combo processing was implemented so that the combo is released on the first key release. But having combo released after the last key has some benefits. Like if your combo resolves to a modifier, it is nice to be able to release some of the fingers so they can be used to type.

And oh boy it would be nice to code in Rust some day. I need more time in my life. I wanna test this repo so badly!

TeXitoi, 2020-10-12:
@sevanteri thanks for the information, that's a very valuable comment!

Do you have any ARM keyboard? Keyberon should be easily portable to any QMK compatible ARM keyboard. If you need any help in this way, feel free to ask.

Or you don't? You might want a https://github.com/TeXitoi/keyseebee, you can source everything yourself if you want. As you've build a gergo, I suppose you have the skills for. Or I can ping you when I have a ready one, and sent it to you?

sevanteri, 2020-10-12:
> Do you have any ARM keyboard? Keyberon should be easily portable to any QMK compatible ARM keyboard. If you need any help in this way, feel free to ask.

I have an Infinity Ergodox that has a MK20DX256 as its controller and it seems to ARM. Never actually looked into what the keyboard has in it as I've moved to smaller keyboards with AVRs.

> Or you don't? You might want a https://github.com/TeXitoi/keyseebee, you can source everything yourself if you want. As you've build a gergo, I suppose you have the skills for. Or I can ping you when I have a ready one, and sent it to you?

Looks really nice, except the thumb cluster seems to small for me :D I like how the bottom is clear of components. Looks really slick!

And no, I've only ever soldered switches and bigger components, never SMD.
Maybe diodes I could handle, but MCUs... nope, way too shaky hands.

TeXitoi, 2020-10-13:
Mmm, I can't find anything on MK20DX256 with rust except old blog post hacking on teensy 3. So it should be doable, but you don't even have a PAC crate, HAL crate, and no USB support. It will be quite a lot of work.

I don't understand your problem with KeySeeBee thumb cluster. Is your problem:
 * 1U thumb button
 * not enough button
 * no outer button

Anyway, if the thumb cluster doesn't fit, better to find something else. There is work in AVR and rust. The only lacking thing today to have keyberon running is a USB device driver: https://github.com/Rahix/avr-hal/issues/40

sevanteri, 2020-10-13:
> It will be quite a lot of work.

And a lot of things I have no experience with at all. :P I think I'll wait a bit and see how things evolve.
 
> I don't understand your problem with KeySeeBee thumb cluster. Is your problem:
>  * 1U thumb button
>  * not enough button
>  * no outer button

No outer buttons. I'm waiting on a Kyria which seems to be pretty perfect for me.

gilescope, 2020-10-15:
As it happens the way I re-worked the release code it would only release on the final key being released - so sounds like that's perfect - thank you very much for the advice! But there is some slight bug in it still that I haven't quite figured out when it happens - because without the combo code the firmware is rock solid and I never have to bounce my keyboard. I second rust firmware - embedded should not be this easy.

TeXitoi, 2020-10-15:
Even if there is bugs, you can open a PR, and explain the bugs you encounter. Another pair of eyes might catch the mistake.

octol, 2020-10-18:
+1 for this functionality! For me this is easily the number one missing feature in the KeySeeBee I recently started using as my daily driver

camrbuss, 2021-11-07:
We should be able to close this issue since #68 was just merged.

TeXitoi, 2021-11-07:
Sure!

closed 2021-11-07 by TeXitoi