TeXitoi/keyberon #37: Tap hold interval

TeXitoi, 2020-11-26:
The field is present in the config (to let the option to release 0.2 without implementing this feature).

We can add a tap_hold_interval corresponding to the maximal time within if you tap and hold the key, it will do the tap action, and stay holded, to allow holding the tapping behavior.

The continuation of #33

jtroo, 2022-04-19:
Ah, I was looking for this feature! A bit sad it's not implemented yet, but glad it's on your radar. I may try my hand at implementing this once my software remapper is in a better state if you haven't gotten there first.

This feature is something that's missing from kmonad that I miss from QMK. I have no idea how to begin contributing to kmonad because it's written in Haskell though, haha.

TeXitoi, 2022-04-19:
I personally don't have the usage of this, as it would not work in my workflow, but I'm open to contributions