TeXitoi/keyberon #40: Teensy LC support?

MarkuBu, 2020-12-24:
Hi, I want to build a keyboard and I want to use Rust instead of QMK. I have some Teensy LC lying around I want to use for the build. 

Does this firmware support Teensy LC?

Regards and merry Christmas

TeXitoi, 2020-12-24:
Keyberon in its own will work anywhere rust works. But, to make a USB keyboard, you'll need embedded hal gpio and USB device implementation.

All I can find for the teensy lc is https://gitlab.com/teensy-rs/teensy-lc/ gpio is present, maybe outdated but it should be easy to update. usb-device is another story.

So, to use a teensy lc, the tricky part will be to have this usb-device implementation.

I'd personally recommend using a WeAct MiniF4: cheap, well done, USBC connector, keyberon example available at https://github.com/TeXitoi/keyberon-f4 Else, almost anything stm32 based will be straintforward.

MarkuBu, 2020-12-24:
Ok, thanks. I think I will use than a Nucleo or Discovery board for my first try.

closed 2020-12-24 by MarkuBu