TeXitoi/keyberon #42: What's the minimal required MCU?

MarkuBu, 2021-01-01:
Hi, I want to make a keyboard PCB and I want to use keyberon as firmware. I have some questions to select a MCU.

What is the minimal required MCU for keyberon? 
How much flash, RAM, clock frequency? 
Is EEPROM useful or not supported at all? 
Is a Cortex-M0 enough?

Happy new year

TeXitoi, 2021-01-01:
STM32 MCUs have good USB support in rust. AVR has not yet USB support in rust, and is less mature.

Firmware is about 22k here, so better 32k or more. I didn't optimized for size, so it should be possible to work a bit to make it fit on 16k.

There is not really eeprom needed.

M0 is enough. I use it on a STM32F072CBT6 without any problem, so 48MHz is enough, maybe less is possible.

MarkuBu, 2021-01-01:
I'm planning to use a STM32L0xx, LQFP32 package. I think EEPROM might become interesting to store some settings

TeXitoi, 2021-01-01:
You'll need a 0x2 or 0x3 to have USB

Not sure for 8k of ram, so I'll opt for a stm32l072kb

MarkuBu, 2021-01-01:
Does it really need so much RAM?

Any experience with ATSAM?

TeXitoi, 2021-01-01:
I don't know, never tested with less than 20k.

TeXitoi, 2021-01-01:
There is https://github.com/atsamd-rs/atsamd/tree/master/hal/src/samd21/usb for USB, didn't tested it personally.

TeXitoi, 2021-01-01:
There is https://github.com/atsamd-rs/atsamd/tree/master/hal/src/samd21/usb and https://github.com/atsamd-rs/atsamd/tree/master/hal/src/common/thumbv7em/usb for USB, didn't tested it personally.

MarkuBu, 2021-01-01:
Thanks, I'm ordering a Nucleo L053R8 and an Adafruit feather board with ATSAMD21G18 to make some tests. I'll let you know the results

MarkuBu, 2021-01-02:
It seems I can't use the  L053. It has USB, but no USB DFU bootloader. The L053 only supports USART bootloader

I think the most simple way to flash a controller is DFU because it is built in

TeXitoi, 2021-01-03:
While a hardware DFU bootloader is a plus, you can always flash using a debugger probe as a ST-Link, that is most of the time included in the nucleo boards.

MarkuBu, 2021-01-04:
> While a hardware DFU bootloader is a plus, you can always flash using a debugger probe as a ST-Link, that is most of the time included in the nucleo boards.

I have a ST-Link, but if someone else wants to build a keyboard it should be possible to program the board without an external programmer.

The problem with the Nucleo boards is that you can't use USB from the main controller. You have to wire a connector yourself.

closed 2021-07-07 by TeXitoi