TeXitoi/keyberon #43: Custom keycodes

gondolyr, 2021-01-04:
I have a [Drop Ctrl keyboard](https://github.com/qmk/qmk_firmware/tree/master/keyboards/massdrop/ctrl) and want to implement the custom key codes like how it is defined in QMK here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/massdrop/ctrl/keymaps/default_md/keymap.c#L3-L22.

Would it be possible to do something like this or should I adjust my approach?

TeXitoi, 2021-01-04:
Do you already have a working firmware on the Drop Ctrl? If yes, I'd like to link to such a firmware, to show a keyberon firmware to this keyboard.

This is named custom action in keyberon. That's on the upcoming v0.2, not on crates.io, but available on master. You define your enum with the variant you want, and then, your layout is of type `Layout<MyCustomActions>`. When you `layout.tick()`, it'll return a custom action if available, and you can act as you want with it.

gondolyr, 2021-01-04:
I don't have a working firmware yet unfortunately. I still have a lot of work to do and translating the [ST-specific code](https://crates.io/crates/stm32f1xx-hal) to work with the [atsamd51 ](https://crates.io/crates/atsamd-hal) MCU is currently my biggest challenge, particularly because there are no public examples of this MCU being used with this library.

For now, I plan on creating a basic firmware without the LED key codes on that layer just so I have something to show. I look forward to the changes in the 0.2 release! 😄

TeXitoi, 2021-01-04:
There is a hal for atsamd51 already, with a usb-device implementation. You should just have some config glue code to do to have a basic keyboard using keyberon

https://crates.io/crates/atsamd-hal

TeXitoi, 2021-01-04:
Mmm, OK, you've got it ;-)

TeXitoi, 2021-01-04:
https://github.com/atsamd-rs/atsamd/blob/master/boards/feather_m4/examples/usb_echo.rs should be a good start to configure the USB. Configuring the gpio should be similar to stm32, and you can also look at blinky examples. If you have any question, just ask.

gondolyr, 2021-01-08:
I've been staring at the Feather M4 example code but I haven't been able to understand what I need to do. I am new to writing embedded devices so I feel way over my head right now. I've tried reading the Discovery book ([link](https://docs.rust-embedded.org/discovery/index.html)) and the ([Embedded Rust book](https://docs.rust-embedded.org/book/intro/index.html)) but found them unsuitable for a beginner for the most part.

I've narrowed down the pins I need for the USB connection to the pins PA24 (D-) and PA25 (D+) but haven't been able to figure out what to do with this so far. I have a WIP commit here: https://gitlab.com/gondolyr/keyberon-drop-ctrl/-/blob/firmware/src/main.rs#L74. I'll still keep reading what I can and try to make progress. 🙂 

Aside: I wasn't certain if I should open a new issue but when I explored recreating the Gergo layout ([QMK link](https://github.com/qmk/qmk_firmware/blob/master/keyboards/gergo/keymaps/germ/keymap.c#L52)), I ran into the issue where the available KeyCode variants don't include the punctuation/symbols directly so mapping the symbols layer doesn't seem possible currently.

TeXitoi, 2021-02-17:
Sorry, missed this comment.

For the punctuation, you'll have to `m(&[LShift, Kb2])` for @ for example. I use a dedicated macro for shifted key here: https://github.com/TeXitoi/keyberon-f4/blob/master/src/main.rs#L105

Your code seems OK, does it compiles?