TeXitoi/keyberon #46: USB not enumerating on stm32f042xx

Skelebot, 2021-01-27:
Mostly out of curiosity I tried to run keyberon on a STM32F042F6P6 (it has enough flash and built-in USB), but the device does not enumerate and UsbDevice.poll always returns false. I constructed a minimal example:
```
#![no_std]
#![no_main]

extern crate panic_halt;

use cortex_m_rt::entry;

use stm32f0xx_hal as hal;
use hal::prelude::*;

use usb_device::class::UsbClass;
use usb_device::bus::UsbBusAllocator;
use hal::usb;

#[entry]
fn main() -> ! {
    let mut p = hal::pac::Peripherals::take().unwrap();
    static mut USB_BUS: Option<UsbBusAllocator<hal::usb::UsbBusType>> = None;

    let mut rcc = p
        .RCC
        .configure()
        .hsi48()
        .enable_crs(p.CRS)
        .sysclk(48.mhz())
        .pclk(24.mhz())
        .freeze(&mut p.FLASH);

    let gpioa = p.GPIOA.split(&mut rcc);

    let usb = usb::Peripheral {
        usb: p.USB,
        pin_dm: gpioa.pa11,
        pin_dp: gpioa.pa12,
    };
    let usb_bus = unsafe {
        USB_BUS = Some(stm32f0xx_hal::usb::UsbBusType::new(usb));
        USB_BUS.as_ref().unwrap()
    };

    let mut usb_class = keyberon::new_class(usb_bus, ());
    let mut usb_dev = keyberon::new_device(usb_bus);
    
    loop {
        if usb_dev.poll(&mut [&mut usb_class]) {
            // This is never reached
            usb_class.poll();
        }
    }
}
```
This code depends directly on the master branch of keyberon. Dmesg -w always shows
```
new full-speed USB device number 55 using xhci_hcd
Device not responding to setup address.
Device not responding to setup address.
device not accepting address 55, error -71
```

TeXitoi, 2021-01-27:
Maybe https://github.com/stm32-rs/stm32f0xx-hal/blob/master/examples/usb_serial.rs#L19

Skelebot, 2021-01-28:
Thank you, I completely forgot about remapping pins. This solved the issue.

closed 2021-01-28 by Skelebot