TeXitoi/keyberon #47: Wakeup Computer from S3 Suspend

camrbuss, 2021-02-17:
I am having difficulty getting my keyboard to wakeup my computer from an S3 suspend. 

I am running Linux Kernel 5.10 and have a custom STM32F072 keyboard and can enter sleep with the `MediaSleep` keycode, but cannot seem to wake my computer with a button press. QMK on the same keyboard does wake up my machine, so I do not believe it is a hardware issue. I also tested with the Keyseebee firmware after just changing the pins, and still was unable to wake my machine. 

Has anyone else had this issue? I looked into the [stm32-usbd](https://github.com/stm32-rs/stm32-usbd/blob/9caa7465e834ae967133d83460a02d16035a84c5/src/bus.rs#L286) repository for `suspend` and `resume` but am not sure if those are even the correct functions to call.

BTW amazing library! I love the new CustomAction functionality for entering DFU.

TeXitoi, 2021-02-17:
Sorry no idea. But I doubt that stm32-usbd suspend and resume are related, I think they are for usb protocol suspend/resume.

camrbuss, 2021-02-21:
I looked into QMK and it seems that it calls `usb_lld_wakeup_host` from chibios which sets the `CNTR` register with the `RESUME` bit for 2 milliseconds (USB standard states 1ms-15ms) and then clears the bit. 

As you mentioned, this is not what the stm32-usbd resume or suspend is doing as that has to do with the `LP_MODE` bit, which I believe has more to do with the device than the host. 

I am thinking this should be a part of stm32-usbd and not keyberon. Should I open an issue there?

What are other keyberon users doing to wake their computer from suspend? Power button?

TeXitoi, 2021-02-22:
If that's dedicated register access, yes it should go in stm32-usbd.

I personally use a laptop, so opening lid wake up the computer.

camrbuss, 2021-02-23:
Created [stm32-usbd issue](https://github.com/stm32-rs/stm32-usbd/issues/28) to capture this.

closed 2021-02-23 by camrbuss