TeXitoi/keyberon #48: Layer modifiers only add/stack and don't work as expected

riskable, 2021-02-18:
My keyboard has two function keys ("Fun" and "More Fun").  I have defined these keys as `l(1)` and `l(2)`.  I've also configured "Fun-More Fun" and "More Fun-Fun" (both held at once) to invoke `l(3)`: Layer 1 has "More Fun" set to `l(3)` and layer 2 has "Fun" set to `l(3)` so if both keys are held down (regardless of order) it should invoke layer 3 but that's not what's happening.  Instead of invoking layer 3, pressing both "Fun" and "More Fun" invokes layer 4.

I believe this is because of how `Layout.current_layer()` (https://github.com/TeXitoi/keyberon/blob/master/src/layout.rs#L396) is defined:

```rust
   fn current_layer(&self) -> usize {
        let mut iter = self.states.iter().filter_map(State::get_layer);
        let mut layer = match iter.next() {
            None => self.default_layer,
            Some(l) => l,
        };
        for l in iter {
            layer += l;
        }
        layer
    }
```

Rather than looking at the layer number specified by `Action::Layer()` it is merely adding the current layer number (say, `1`) to the layer defined for that key (say, `3` which would result in `4`).  The test case I've come up with is this:

* Define two keys: One as `l(1)` and one as `l(2)`.
* Define both those keys to go to `l(3)` when the other is pressed.
* Press both at the same time and it behaves as if you defined them as `l(4)` instead of `l(3)`.

I haven't figured out how to fix this yet otherwise I'd submit a PR :smile: 

BTW: I discovered this when I was trying to define "LAlt-Fun-Up" (LAlt configured as `l(4)` in layer 1) to send a thumbs up sequence (👍) on layer 4 while having "RAlt-More Fun-Up" (LAlt configured as `l(5)` in layer 2) send a top arrow (🔝) on layer 5.  It just wasn't working right.

TeXitoi, 2021-02-18:
It's as documented: layers adding.

TeXitoi, 2021-02-18:
For your usecase, you just have to let the layer keys transparent in the other layers, and you'll have your behavior: when pressing layer 1 and layer 2,  you'll have layer 3

riskable, 2021-02-18:
So this is how it's supposed to work?  That's very confusing.  Assuming I've got two layer modifier keys (say, "Fun" for layer 1 and "Fun2" for layer 2) how am I supposed to access layer 3?  If I press "Fun" I get layer 1, if I press "Fun2" I get layer 2.  If I press "Fun" and "Fun2" at the same time I'm getting layer 4.  I suppose I could just leave layer 3 as an always-ignored layer but then things can get ridiculously complicated when trying to define any sort of multi-key shortcut beyond two layer modifier keys.

Say I want to define "Fun-LAlt-Super-Up" (left side modifiers) to send `k(F13)` but "Fun2-RAlt-Apolication-Up" (right side equivalent) to send `k(F14)`.  With the "layers add" method you couldn't do this without doing some complicated math and having a whole bunch of unused layers.  Assuming the left-side shortcut was Fun (`l(1)`), LAlt (`l(3)`, when combined with Fun), Super (`l(4)`, when combined with Fun and LAlt) you'd end up on layer 8 (I *think*).  Now let's assume the right-side equivalent is Fun2 (`l(2)`), RAlt (`l(5)`), Application (`l(6)`) you'd end up on layer 13.

Doing it the "layers add" way you can wind up with the same action being sent for two completely different keyboard shortcuts--with no way to differentiate between them (aka conflicts).  For example, "Fun-LAlt-Super-Up" (layer 8) would end up the same as Fun-Application-Up (also layer 8).  That...  Doesn't make any logical sense from, "I'm defining unique shortcuts for things" standpoint.

TeXitoi, 2021-02-18:
```rust
#[rustfmt::skip]
pub static LAYERS: keyberon::layout::Layers = &[
    &[[l(1),  l(2),  k(A)]],
    &[[Trans, Trans, k(B)]],
    &[[Trans, Trans, k(C)]],
    &[[Trans, Trans, k(D)]],
];
```

* pushing 3rd key will give A
* pushing 1st key then 3rd key gives B
* pushing 2nd key then 3rd key gives C
* pushing 1st and 2nd keys (in any order) and then 3rd key gives D

TeXitoi, 2021-02-18:
So, the basic idea:
 * put layers on your base layer, and transparent in the other layers
 * gives 1, 2, 4, 8... as layers keys
 * the layers sum, so I want layer 5, I press 4+1 and I have it

riskable, 2021-02-18:
In your example layer 3 is unreachable:

```rust
#[rustfmt::skip]
pub static LAYERS: keyberon::layout::Layers = &[
    &[[l(1),  l(2),  k(A)]],
    &[[Trans, Trans, k(B)]],
    &[[Trans, Trans, k(C)]],
    &[[Trans, Trans, k(D)]],
];
```
You can get `l(1)` and `l(2)` but pressing `l(1)` + `l(2)` goes straight to layer 4.  This concept also still has the same problem I was talking about with conflicting shortcuts.

The real problem (to me) is that defining your layer-based shortcuts this way becomes extra hard/complicated ("OK I want a shortcut that involves 3 modifiers that I can mash all at once...  Let's do some math! ...and hope I don't have conflicts later").  Instead of simply defining the `l(2)` key as going to layer 3 inside layer 1 you now have to think about how multiple modifiers add up together to figure out which layer they'll end up on.  You basically have to chart it all out before you start messing with `layers.rs`...

"OK LAlt adds 4 to the current layer when layer 1 is active.  That makes 5.  When Super is held while layer 5 is active it adds...  Let me check.  OK it's `l(5)`.  So that means I have to modify layer 10.  <starts counting array items inside `layers.rs`>."  Now I also have to worry about if any other combination of modifiers adds up to 10 and double-check I don't have anything defined for that already.  I just went through this and IMHO:

* It's a real pain to define anything beyond two or three layers.
* I have to think way too hard about what adds up to what to mentally map any given more-than-two-modifiers shortcut.
* It doesn't make logical sense that two different completely different keystrokes would result in the same shortcut (unless I defined them individually to be the same).
* It results in a *much* larger `Layout` than would be necessary if you didn't have layer adding.

riskable, 2021-02-18:
Another way to think about this problem:  Let's say I want to make a user-friendly interface for configuring Keyberon.  I want the user to be able to press their desired keystroke (e.g. Fun-LAlt-Super-Up).  I record that keystroke and now need to generate a `Layout` based on that.  How do I do it?  Well, I need to define "Fun" as its own layer.  Say, layer 1.  Next we need to define LAlt as its own layer--but only when Fun is held.  We'll use `l(3)` (because Fun2 has `l(2)`).  Now we do the same for Super so it gets `l(4)`.  Also, for good measure we define LAlt to be `l(3)` when Super is held (just in case the user presses those two a bit out-of-order--they still need to press Fun before anything else though).

Assuming there's already a default layer we generate full (mostly-empty) arrays for layers 1-7 (0 indexed) and put our action in the appropriate position.

Now the user records a number of other keystrokes and one of them adds up to the same position in layer 8.  What do we do in that situation?  The user provided two completely different keystrokes but we can't bind them to two different actions because the ultimate position is the same.

There's a workaround of course:  Give each layer some large, odd number like 21, 31, 41, etc but then we end up with a *really* big `Layout` that is 99% empty slots.  It doesn't seem logical to me when you can emulate addable layers--if you want--by just being studious in the way you define your `Layout`.

For example, if Keyberon didn't add layers like it does you could still define Fun as `l(1)` and Fun2 as (`l2`) and have them reach layer 4 together by simply defining Fun as `l(4)` inside of layer 2 and Fun2 as `l(4)` inside of layer 1.

riskable, 2021-02-18:
I just realized I've got the wrong idea about your example...  You're saying to leave the Fun and Fun2 modifiers as `Trans` on subsequent layers and if you do that you can reach layer 3.  I get it now.  You can ignore my statement about layer 3 being unreachable. It's really layer 4 that's unreachable.

Now I'm even _more_ confused as to how to properly define multi-modifier shortcuts.  Sigh.

riskable, 2021-02-18:
So I've been thinking about how you told me to do this...  To define something like "Fun-LAlt-Super-Up" I'd have to do it like so:

* Fun and Fun2 are `l(1)` and `l(2)` but `Trans` on all other layers
* On layer 3 I define LAlt as `l(1)` to reach layer 4 (this is the part that's super confusing--in my head it's much more logical to define LAlt as `l(4)` to get to layer 4...  But no, that would go to layer 7).
* On layer 4 I also define Super as `l(1)` to get to layer 5.

Now if I want to bind a shortcut to "Fun2-RAlt-Application-Up"...

* Fun and Fun2 are `l(1)` and `l(2)` but `Trans` on all other layers (same as before)
* On layer 3 I define RAlt as `l(1)` to reach layer 4.
* On layer 4 I also define Application as `l(1)` to get to layer 5.

So even though these shortcuts are completely different they end up in the same exact position in the `Layout`.  Alternatively, I could define RAlt or Application as `l(2)` which would let me put the shortcut on layer 6.  But this too still could conflict with other keyboard shortcuts.  I have to be *very careful* with my `Layout` using this method or there will be conflicts.

The part of this that doesn't make sense to me is that the method of adding layers adds an unnecessary layer (haha, get it?) of complexity to defining your `Layout`.  Defining shortcuts becomes like bitbanging binary instead of just "follow the function".

TeXitoi, 2021-02-18:
First, I don't pretend my solution is the good one. When I've done it, it was for me simpler to implement and to configure: I don't have to be cautious: several function keys, whatever order is pressed, will give the same layer.

Using only power of 2 as layer keys mean each combo will give a different layer. You can see that as puting a 0 or a 1 at the bit position.

Now, it doesn't mix well with "layer key not always present" I agree.

Some inconvenient on your solution
 * You must be very cautious that different paths of the same layer modifiers gives the same layer
 * You have a more complicated layout for the classic "lower" "raise" "combo" 4 layers with 2 fn.
 * breaking change
 * different behaviors when you release layer keys (I don't really know the more intuitive behavior)

If you want to experiment, I think this patch will do the trick:

```diff
diff --git a/src/layout.rs b/src/layout.rs
index a3719ad..dc3fa0e 100644
--- a/src/layout.rs
+++ b/src/layout.rs
@@ -400,7 +400,7 @@ impl<T: 'static> Layout<T> {
             Some(l) => l,
         };
         for l in iter {
-            layer += l;
+            layer = l;
         }
         layer
     }
```

A cleaner fix would be needed for integration, but like that you can experiment, and give feedbacks.

That would be a breaking change.

riskable, 2021-02-18:
Well *I* don't care about breaking changes haha.  I'm more than happy to change all my stuff around if it means an overall better solution 👍.

Also: Your change to `layout.rs` works *awesome*!  OMG that's exactly how I was expecting it to work.  That way makes so much more sense to me from a configuration standpoint.  No need to do math to figure out where my keystrokes are going to wind up :smile: 

...but I'm with you for the most part:  "What's a better way?"  I mean, I can think of better ways from an end user perspective but from a data structure standpoint I would *prefer* my way (because my brain sucks at calculating bits) but your way is probably fine if having loads of (mostly empty) layers doesn't add much to the stack/memory footprint.

I've been re-jiggering my Riskeyboard 70 code quite a lot lately and one of the things I've been working on is configuration.  Having a file in `src/` that end users need to configure is a big no-no and worst-practices from a Rust ecosystem standpoint.  I've already moved all my firmware configurables (actuation/release thresholds, LED stuff, display stuff, etc) into a `Config.toml` that gets converted into a `config.rs` at compile time (via `build.rs`) and now I'm trying to come up with a better way to handle the actual *key* (and rotary encoder) configuration.

Using a .toml is the lowest friction route but I'm not opposed to JSON5 or a similar "modern" format.  Whatever I choose I will share publicly and I'd like to have *some* sort of configuration tool available as its own crate.  Preferably something that can work with anyone's Keyberon-powered keyboard.

commit 5e2ae6e5 (2021-10-18): Change Layer action to more intuitive "last hold" behavior.

Fixes #48
closed 2021-10-18 by TeXitoi