TeXitoi/keyberon #51: Rows as Input and Cols as Output

Cimex97, 2021-04-05:
Hey it is possible to use the Cols as Output instead of inputs?

TeXitoi, 2021-04-05:
You can just swap the 2 coordinates before sending it to the layout.

Cimex97, 2021-04-05:
In which file I need to do it?

TeXitoi, 2021-04-05:
I don't know your firmware, but something like that: https://github.com/TeXitoi/keyseebee/blob/4d29baf283a676035740669f2bfbe211ac492998/firmware/src/main.rs#L271

Transform would be something like `|e| e.transform(|i, j| (j, i))`

Cimex97, 2021-04-05:
It is the base of your ortho60-keyberon project.

I need to define my Cols like
```c
pub struct Cols(
    pub PA0<Output<PushPull>>,
    pub PA1<Output<PushPull>>,
    pub PA2<Output<PushPull>>,
    pub PA3<Output<PushPull>>,
    pub PA4<Output<PushPull>>,
    pub PA5<Output<PushPull>>,
    pub PA6<Output<PushPull>>,
);

```
But I got this error: 
````
let matrix = Matrix::new(
^^^^^^^^^^^ expected trait `embedded_hal::digital::v2::OutputPin`, found trait `embedded_hal::digital::v2::InputPin`
````
This error message looks for me that I olny can define Cols as InputPins and Rows as OutputPins

Sorry I'm a rust newbie.

Cimex97, 2021-04-05:
https://github.com/TeXitoi/keyberon/blob/653f5d9b5d39994becb3125d44bfadc4f7db5096/src/matrix.rs#L48
In your Matrix implementation you used the Col as OutputPin and Row as InputPin. So maybe it is not a Row or a Col for me, just an Input and Output for "scanning" purpose.

So if I switch my Cols and Rows I can compile the firmware. Also I transform the coordinates with your suggested function that.
But now I don't get any key output. 
Do you have an idea how I can debug the software?

TeXitoi, 2021-04-05:
That's the trick yeah.

Maybe diodes are in the wrong way? Try shorting some input and output without diode. Also with a debugger you can use gdb do go step by step.

Cimex97, 2021-04-05:
Ah well, I found the Problem. My hardware works with a PullDown instead of a PullUp for the Inputs. Maybe it is a good idea to specify which is the Input or Output and if the Input works with a PullUp or PullDown while create the matrix? I can try do to this and create a Pull Request. But this will take some time.

BTW: Thank you for your very fast support :)

TeXitoi, 2021-04-05:
In this case, the default row is output should work.

Cimex97, 2021-04-05:
If I use the row as output I got sometime more than one keycode output. 

Pin1 of the Cherry-MX key is connected to a Col. 
Pin2 ot the Cherry-MX key is connected to the row with a diode.

[image]

So I think I can't use the rows as output. (I'm not a hardware expert at all)

TeXitoi, 2021-04-05:
That's the standard row output active high, so that's implemented by default.

Don't you have any short or any diode in the wrong direction? If you have several keypress while pushing only one button, I'll check for shorts looking at the keys that activate.

TeXitoi, 2021-04-05:
What's tour PCB? Does it works flawlessly with the default firmware? If that's some normal keyboard, maybe some keys activates several intersections, and that's by design, do you have the complete schema?

Cimex97, 2021-04-05:
It is a custom PCB. The matrix have 10 rows and 7 cols.
This here is the left side of the keyboard. I need here the output on the cols (at top in this image) and the input to the rows (left in this image). 
[image]

All diodes are in the same direction. I checked this after soldering.

TeXitoi, 2021-04-05:
Are the resistor in series or is it a pull up (to 3.3v) or a pull down (to ground)?

Cimex97, 2021-04-05:
In series.

TeXitoi, 2021-04-05:
Should not cause any problem.

TeXitoi, 2021-04-05:
I'd check the keys that double register and try to find a pattern related to the col and row number. You might have a short between 2 cols, that would explain the double register.

Cimex97, 2021-04-05:
I will clarify this and get back to you. Thanks for your help.

Cimex97, 2021-04-08:
You were right, but there does not seem to be a short. Because when I swap the affected row with another, the problem shifts to that row.
Could it be because I don't have my rows from 1-10 as in the hardware, but 1,6,2,7,3,8,4,9,5,10? This is because the left side of the keyboard is row 1-5 and the right side is 6-10. 
I have now adjusted this so that I have line 1-10 and it works fine.

If you are not interested in the feature, the issue can be closed.

TeXitoi, 2021-04-08:
So, that's 2 keys from same colum, successive rows? or 2 rows whatever the order, always the same pins?

Cimex97, 2021-04-09:
I change only the pins on the controller side. The rows and cols on the matrix are fixed, because it is a PCB.
If I press my Col1(Pa1) Row3 switch and Col2(Pa2) Row4 the Col3(Pa1) Row 4 is recognized. 
I change now the Pins of the Controller from
Pin on Controller => Pin on Matrix 
- Pa1 => Col1 
- Pa2 => Col2 
- Pa3 => Col3 
- Pa4 => Col4 
- Pa5 => Col5 
- Pa6 => Col6 
- Pa7 => Col7 
to
Pin on Controller => Pin on Matrix 
- Pa3 => Col1 
- Pa4 => Col2 
- Pa5 => Col3 
- Pa6 => Col4 
- Pa7 => Col5 
- Pa1 => Col6 
- Pa2 => Col7 

Then the double key press is recognized at Col 6 and 7. I also switch the pins to PB0 and PB1 and it is the same.  Switching the rows has also the same effect.

I have found a workaround to "fix" it.
I activate with (Col1Row3) my layer(2) and on this layer I set  Col1Row4 to NoOp.

TeXitoi, 2021-04-09:
Do you use PB2 with a WeAct MiniF4? https://github.com/WeActTC/MiniSTM32F4x1

PB2 has a pull up on the board, and it can mess things. Better to not use this pin.

In your message, you said that Col1 and Col3 use Pa1, I suppose this is a typo. If the 2 columns use the same pin, it can't work properly.

Cimex97, 2021-04-09:
Yes it is a typo :D
I use the stm32f103c8t6 Bluepill

TeXitoi, 2021-04-09:
It really looks like a short, a diode in the wrong direction or a faulty diode. The fact that it's more corrected to the board pin than to the button make me think of something on the board. Can you isolate the 2 cols and 2 rows that case the problem? do you reproduce the problem by emulating the button by touching the pins with 2 diodes in the good direction? can you give me the pin, I might find the time to test here on a bluepill.

Cimex97, 2021-04-09:
I will test it again with your tips and get back to you. Also I will try another bluepill.

Cimex97, 2021-04-09:
These are my pins for the matrix:
```c
pub struct Cols(
    pub PA0<Input<PullUp>>,
    pub PA1<Input<PullUp>>,
    pub PA2<Input<PullUp>>,
    pub PA3<Input<PullUp>>,
    pub PA4<Input<PullUp>>,
    pub PA5<Input<PullUp>>,
    pub PA6<Input<PullUp>>,
);

pub struct Rows(
    pub PA9<Output<PushPull>>, // 1
    pub PA10<Output<PushPull>>, // 2
    pub PA15<Output<PushPull>>, // 3
    pub PB3<Output<PushPull>>, // 4
    pub PB4<Output<PushPull>>, // 5
    pub PB5<Output<PushPull>>, // 6
    pub PB6<Output<PushPull>>, // 7
    pub PB7<Output<PushPull>>, // 8
    pub PB8<Output<PushPull>>, // 9
    pub PB9<Output<PushPull>>, // 10
);
```

TeXitoi, 2021-04-09:
PA15, PB3 and PB4? I suppose you deactivated the debugger functionalities on those pins? Maybe these is some bugs around pullups that are wrongly activated with the remapping? You may try to transform them in input pullup, input floating and finally in output push pull to be sure the configuration is correct.

Cimex97, 2021-06-07:
If you want you can take a look at my [layout.rs](https://github.com/Cimex97/ortho60-keyberon/blob/master/src/layout.rs) and [main.rs](https://github.com/Cimex97/ortho60-keyberon/blob/master/src/main.rs)