TeXitoi/keyberon #57: Add support for media buttons

peauters, 2021-05-05:
At the moment Keyberon has support for key codes in the Generic Desktop Ctrls usage page. This includes the key codes up to and including `RGui` which has value `0xE7`.

There is support in the HID spec for some more controls that might be useful to add that are in the `Consumer Control` page.
These could include [pulled from QMK](https://github.com/qmk/qmk_firmware/blob/9dea6f772077dc5c09daf40378e45884d29ab2e2/tmk_core/common/report.h):
```
// 15.5 Display Controls
    SNAPSHOT               = 0x065,
    BRIGHTNESS_UP          = 0x06F, // https://www.usb.org/sites/default/files/hutrr41_0.pdf
    BRIGHTNESS_DOWN        = 0x070,
    // 15.7 Transport Controls
    TRANSPORT_RECORD       = 0x0B2,
    TRANSPORT_FAST_FORWARD = 0x0B3,
    TRANSPORT_REWIND       = 0x0B4,
    TRANSPORT_NEXT_TRACK   = 0x0B5,
    TRANSPORT_PREV_TRACK   = 0x0B6,
    TRANSPORT_STOP         = 0x0B7,
    TRANSPORT_EJECT        = 0x0B8,
    TRANSPORT_RANDOM_PLAY  = 0x0B9,
    TRANSPORT_STOP_EJECT   = 0x0CC,
    TRANSPORT_PLAY_PAUSE   = 0x0CD,
    // 15.9.1 Audio Controls - Volume
    AUDIO_MUTE             = 0x0E2,
    AUDIO_VOL_UP           = 0x0E9,
    AUDIO_VOL_DOWN         = 0x0EA,
    // 15.15 Application Launch Buttons
    AL_CC_CONFIG           = 0x183,
    AL_EMAIL               = 0x18A,
    AL_CALCULATOR          = 0x192,
    AL_LOCAL_BROWSER       = 0x194,
    AL_LOCK                = 0x19E,
    AL_CONTROL_PANEL       = 0x19F,
    AL_ASSISTANT           = 0x1CB,
    AL_KEYBOARD_LAYOUT     = 0x1AE,
    // 15.16 Generic GUI Application Controls
    AC_MINIMIZE            = 0x206,
    AC_SEARCH              = 0x221,
    AC_HOME                = 0x223,
    AC_BACK                = 0x224,
    AC_FORWARD             = 0x225,
    AC_STOP                = 0x226,
    AC_REFRESH             = 0x227,
    AC_BOOKMARKS           = 0x22A
```

I'm opening this issue to start a discussion on if/how this could be implemented.

In order to implement this, the report descriptor would need to be updated to include a second report on the Consumer Control page. This would mean that report IDs would need to be introduced and key codes routed to the correct report and multiple reports would need to be sent via the usb class. 

I was thinking that KbHidReport could be extended to store two reports internally and then write one or two reports as bytes depending on what keys have been pressed.

What do you think?

Skelebot, 2021-05-05:
I think this is a very important feature. If we extended keyberon to be able to use multiple reports, we could also add controls like mouse movement or absolute volume (think rotary encoders or potentiometers on keyboards) which are present in other firmwares like QMK. If I find some time this week I might look into this.

TeXitoi, 2021-05-05:
Well, i'd like to support media keys, but not like that. Mouse is also out of scope of keyberon, but custom action is explicitly here for this.

A design have to be proposed. I'd like that a only media keys device can be implemented, in a flexible and composable way.

peauters, 2021-05-05:
I'm going to work on something tonight. 

What are your thoughts on using features for this?

TeXitoi, 2021-05-05:
No feature, different object that can be combined. Something like a dedicated enum for the keys, a dedicated object for the endpoint, and some helpers for easy double endpoints.

peauters, 2021-05-05:
How do you think it should interact with the Layout? 

Are you thinking multiple USB devices?

As an aside, this is my first major rust project so please do feel free to give more pointers =o)

TeXitoi, 2021-05-05:
For a first draft, you can prototype outside keyberon: define an enum with the multimedia keys, define your USB device, and use Action::Custom to reuse the layout. Then, we can see how to facilitate the interface, but the hardware part will be ready. Refactoring in rust is great.

peauters, 2021-05-06:
So I have it working.

I tried to define a separate UsbClass and have it as a separate endpoint, but I couldn't get that working.

What ended up doing was creating a UsbClass that had dual reports and manage sending the different reports on an event. 

I used CustomActions and my own enum to define the MediaKeys and then reported a MediaKey report if the result of tick() was one of the actions.

I had to make one change to Keyberon to facilitate this and that was to add `max_packet_length()` to the HidDevice trait so that I could increase the length to 9 for the keyboard report with the first byte being the report id.

Code can be found here: https://github.com/peauters/peautkb. Would love comments and thoughts.

peauters, 2021-05-07:
Just revisiting this.

Would you be happy to merge a change to allow different max_packet_lengths to be defined?

TeXitoi, 2021-05-07:
I'm not sure what it means, better to show me the diff

peauters, 2021-05-07:
When allocating the endpoint for the HidClass, you have to specify a max packet size. This was originally fixed at 8, which for the report format in Keyberon is perfect. I needed to increase it for the report format I created.

```diff
     pub fn new(device: D, alloc: &UsbBusAllocator<B>) -> HidClass<'_, B, D> {
+        let max_packet_size = device.max_packet_size();
         HidClass {
             device,
             interface: alloc.interface(),
-            endpoint_interrupt_in: alloc.interrupt(8, 10),
+            endpoint_interrupt_in: alloc.interrupt(max_packet_size, 10),
             expect_interrupt_in_complete: false,
         }
     }
```

I put a `max_packet_size()` on the HidDevice and implemented it as 8 for Keyberon `Keyboard`

```diff
 pub trait HidDevice {

     fn protocol(&self) -> Protocol;

+    fn max_packet_size(&self) -> u16;
+
     fn report_descriptor(&self) -> &[u8];

     fn set_report(&mut self, report_type: ReportType, report_id: u8, data: &[u8])
```

TeXitoi, 2021-05-07:
OK for HidDevice.

TeXitoi, 2021-05-07:
Or not, I'm not sure.

peauters, 2021-05-07:
Will close this as it's been possible without changes to keyberon =o)

closed 2021-05-07 by peauters