TeXitoi/keyberon #60: Problems with the `impl_heteroenous_array` macro

MangoIV, 2021-05-18:
Hi! I'm trying to get my keyboard based on a SAM21 MCU up and running on keyberon. When declaring Pins I've run into an issue with the `impl_heteroenous_array` macro as it seems to expect exactly the pin declaration as implemented in the stm32f0xx_hal. When I do a pin declaration as it's possible with the atsamd_hal:
```rust
pub struct Cols(
    Pin<v2::PA02,Input<PullUp>>,
    Pin<v2::PA04,Input<PullUp>>,
    Pin<v2::PA10,Input<PullUp>>,
    Pin<v2::PA11,Input<PullUp>>,
    Pin<v2::PA09,Input<PullUp>>,
);
```
The macro doesn't work for that. The errors that are thrown tell me that some traits are not implemented:
```error[E0277]: the trait bound atsamd_hal::gpio::v2::Pin<PA02, atsamd_hal::gpio::v2::Input<atsamd_hal::gpio::v2::PullUp>>: _embedded_hal_digital_InputPin is not satisfied```

Strangely it works with the implementation of the rows which is *exactly the same*:
```rust
pub struct Rows(
    Pin<v2::PA06,Output<PushPull>>,
    Pin<v2::PA05,Output<PushPull>>,
    Pin<v2::PA07,Output<PushPull>>,
    Pin<v2::PA08,Output<PushPull>>,
);
impl_heterogenous_array! {
    Rows,
    dyn OutputPin<Error = Infallible>,
    U4,
    [0, 1, 2, 3]
}
```
   
This is very confusing for me and I'd appreciate some help with it. I'm really new with rust and I cannot for the life of me figure out how to change my declarations to be accepted by the macro. If there's another method to do the same that the macro does just for my use case but *without* the macro that would be enough already, too. 
Thanks in advance

TeXitoi, 2021-05-18:
Can you push somewhere the code?

TeXitoi, 2021-05-18:
I think you have to use the unproven feature in your hal crate to have InputPin

https://github.com/atsamd-rs/atsamd/blob/master/hal/src/gpio/v2/pin.rs#L139

MangoIV, 2021-05-18:
I will try the unproven feature thingy and check back if it's still not working and supply you with the github repo. Thanks for your help already :)

MangoIV, 2021-05-18:
I think that worked.

closed 2021-05-18 by MangoIV