TeXitoi/keyberon #61: Support for home row mods?

MarkuBu, 2021-06-07:
Is it possible to create home row mods? What I mean is alt, Ctrl, GUI and shift on asdf and jkl;

I know that tap-hold is supported, but that doesn't work for home row mods. Just like in qmk you need to add some special funktions to make it work.

TeXitoi, 2021-06-07:
The default behavior in v0.2 is usable for homerow mod, it is the equivalent of IGNORE_MODTAP_INTERRUPT in QMK.

The master branch has all the feature QMK has (to my knowleadge), and even more, i.e. no constraints on the mod and tap action. You can have a "shift+t" on tap if you want. If I remember correctly, on QMK, you can only have raw keycodes.

TeXitoi, 2021-06-07:
https://github.com/TeXitoi/keyberon/blob/master/src/action.rs#L8-L28 for the HoldTap configuration on the master branch.

MarkuBu, 2021-06-07:
Sounds great

TeXitoi, 2021-06-07:
Just by curiosity, which keyboard are you planning to use/port to keyberon?

MarkuBu, 2021-06-07:
I have an Atreus 42. I want to replace the Teeny with a STM32.

I also want to build somethin like the MiniDox. 5x3 matrix with three thumb buttons

TeXitoi, 2021-06-07:
Great! If you have any question, just open an issue!

closed 2021-06-07 by TeXitoi