TeXitoi/keyberon #63: Is the Ferris 0.2 supported by keyberon?

MarkuBu, 2021-06-27:
I want to build a keyboard like the Ferris and I want to be able to use keyberon and QMK as firmware. As far as I know the Ferris has QMK support. 

The Ferris 2.0 uses a MCP23017 IO expander. Is this supported by keyberon? The only split keyboard with keyberon support is keyseebee, but this has no QMK support as far as I know.

TeXitoi, 2021-06-27:
AFAIK, no one ported yet ferris 0.2 to keyberon, but it should be doable.

TeXitoi, 2021-06-27:
I will try to answer any question if you want to port it yourself. There is a crate for MCP23017 so it should not be too difficult.

MarkuBu, 2021-06-27:
Ok, I'm working on the schematic. I will try to make the board flexible. The original Ferris has two PCBs for each side. Same for the keyseebee. I want to have a single PCB that I can use for both sides. Maybe I am able to make two options. First option is ARM on one side and MCP on the other side (like the Ferris), second option is ARM on both sides (like the keyseebee). Should be doable.

Thanks for your support

MarkuBu, 2021-06-27:
But there is a big problem: STM32 are sold out, almost everywhere. At least in Europe. Maybe I should add a ATmega32U4 option?

TeXitoi, 2021-06-27:
There is no USB implementation on atmega32u4 yet. Also avr rust support is not great. So you'll have to forget rust on it.

I think nrf52 is usable in rust, bit you don't have yet BT support AFAIK. nrf52 is not supported by QMK, bit you have other promising project, as ZMK.

closed 2021-07-07 by TeXitoi