TeXitoi/keyberon #69: v0.2.0 Release?

haata, 2021-10-15:
heapless 0.7 is nice, would you mind pushing a new release for 0.2.0?

camrbuss, 2021-12-19:
It also might be good to squeeze in embedded hal 1.0 support for the 0.2 release.

TeXitoi, 2021-12-19:
Yeah, and the future default type parameters with const.

TeXitoi, 2022-04-19:
I think we have everything for a 0.2 release. So please test the latest master and give some feedbacks before I decide to release it!

daehyeok, 2023-06-05:
> embedded hal 1 is not out, but we might wait a lot of time to have 1.0 released.

Do you have plan to release another 0.1.x? include chord many feature seems missed on last release.

TeXitoi, 2023-06-05:
Most of the users live with a got dependency. But as there is not so much evolution, and embedded hal 1.0 isn't released, I think we can release a 0.2 as is.

Ben-PH, 2023-07-21:
+1 on the release. I'm currently using a a forked repo as the dependency, and love to be depending on a crates release.