TeXitoi/keyberon #70: Best method for keeping track of binary repos using keyberon for popular keyboards

camrbuss, 2021-11-07:
What is the best way the keyberon "community" can keep track of keyboard firmware that has been put together for common keyboards?

If the designer of the keyboard supports QMK, but someone else writes a keyberon firmware for the board, it would be nice to keep track of this. 

It seems like most popular keyboards running a Arm Cortex-M MCU could be running keyberon, and it would be nice to share this work with newcomers. 

I see the readme starts to do this, but that could get really unwieldy.

TeXitoi, 2021-11-07:
A simple solution would be a dedicated markdown file with some kind of table listing any keyboard and/or firmware?

camrbuss, 2021-11-07:
I like that idea. Simple enough and would just take a PR from someone with an addition/modification.

What are you thinking `doc/KEYBOARDS.md`, `KEYBOARDS.md`, `BOARDIMPLEMENTATIONS.md`, `BOARDS.md`, or `IMPLEMENTATIONS.md`?

I am thinking a nice table with the following columns.

| Keyboard         | PCB or Handwired | Firmware License | Feature Status |
|------------------|------------------|------------------|----------------|
| [KeySeeBee](https://github.com/TeXitoi/keyseebee)        | PCB              | MIT              | <ul><li>- [x] Matrix </li><li>- [x] Split</li></ul>    |
| [Arisu](https://github.com/help-14/arisu-handwired)            | Handwired              | MIT              | <ul><li>- [x] Matrix </li></ul>         |
| <name> with link | <>               | <>               |        <ul><li>- [x] Matrix </li><li>- [ ] OLED</li></ul>      |

DrewTChrist, 2022-01-25:
I'd like to second this. I've recently just developed a firmware with Keyberon for an RP2040 keyboard. The only RP2040 examples I know of belong to @camrbuss and I only happened to find them by searching the around on the internet. I think a list of known Keyberon firmwares would help those wishing to write their own.

PR #75 (2022-02-01): Add table to show repositories using keyberon
Would resolve #70
closed 2022-02-01 by TeXitoi