TeXitoi/keyberon #72: Release shift action

wahtak, 2021-12-04:
My use case is to have `\` and `/` on a single key. 
To get a `\` I want to press this key without shift.
To get a `/` I want to press this key while holding shift.

`/` is non-shifted on the US layout and thus this would require multiple actions:
- temporally releasing shift
- press `/` key
- holding shift again to recreate the correct state

Looking at the code, I think I need a new `Action` which creates a `CustomEvent::Release`? I am pretty lost though, due to my unfamiliarity with Rust. Happy to dig in deeper with your guidance, though.

camrbuss, 2021-12-19:
Instead of using a "Shift" key, why not make that key a new layer and shift everything on the layer except for the `\` and `/`?

TeXitoi, 2022-04-19:
Keyberon don't allow you to modify all the state of the keyboard, so it is basically impossible (now) to release a pushed key. Such a feature is difficult to do cleanly and might produce a lot of subtil bugs.

So, if something like that is really needed, we need to be careful to design something well done.

The layer trick should be a good workaround.