TeXitoi/keyberon #73: implementing new keyboard

alinelena, 2021-12-21:
Hi,

i try to implement a new keyboard in keyberon is a 5x13 with an encoder and few led and a strip.
At the moment all is working in qmk.
here is what I did 
https://github.com/m-lego/m65
it is practically a copy paste clean from keyberon-f4

now questions I have.
1. are encoders supported?
2. are led strips supported?
3. is there any contribution guide?
4. can i implement qmk tap dance? now i have two taps toggle the layer permanently 0/x and puts a led on/off to indicate which layer is active 
5. utf-8 support?
now issues i have... I did not test extensively my keyberon firmware but one issue I noticed is
shift + a gives me A| which indicates to me something nefarious with the pins I use. but I do not see that in qmk

TeXitoi, 2021-12-22:
First, keyberon is a lib dedicated to keyboard, not a framework. So you have the building blocks to do a keyboard, and you can combine with other crates for the rest.

Encoders are not directly supported, but it is as easy as sending a press and a release of some coords when an event came from the encoder, so quite easy.

Same for led strip. You can add custom event on the action type, allowing to manage anything from a keypress, as on, off, change mode/color/whatever.

There is no contribution guide. If you miss a feature, open an issue, and we will design the interface of the feature, and then you can implement it (or let someone else do it if you don't have the time or change priority in your life).

For something like tap dance, please open a dedicated issue. I'm not aware of this exact feature, but will help designing something with pleasure. Same for utf8.

TeXitoi, 2021-12-22:
For your problem, it.looks like goshting, but it might be some capacitance on your PCB, adding some timing around the activation of the row might help. I never needed such timing, but maybe you need it.

closed 2022-04-19 by TeXitoi