TeXitoi/keyberon #81: keyberon without RTIC

romen, 2022-03-03:
Is there any example of using Keyberon without RTIC?

I am trying to put together a keypad using a Longan Nano board which features a RISC-V MCU.
RTIC at the moment is not supported outside of `cortex` MCUs, so I cannot use it.

In particular I am trying to figure out how to replace

~~~rust
fn send_report(iter: impl Iterator<Item = KeyCode>, usb_class: &mut resources::usb_class<'_>) {
    use rtic::Mutex;
    let report: KbHidReport = iter.collect();
    if usb_class.lock(|k| k.device_mut().set_keyboard_report(report.clone())) {
        while let Ok(0) = usb_class.lock(|k| k.write(report.as_bytes())) {}
    }
}
~~~

TeXitoi, 2022-03-03:
I don't know any implementation without RTIC, bit there is no dependances on RTIC so it is doable.

The particular line is called when an even occurs. You'll also have to poll the USB device at least each 5ms by memory.

I can review your code of you have some non working draft (or even a working one if you want).

camrbuss, 2022-03-03:
Before I had RTIC working for the RP2040, I had the keyboard working with just `main()`.

From a early commit in the repo:
https://github.com/camrbuss/pinci/blob/70acdeaf52759e7c79465029ee994d48295f430c/firmware/src/main.rs

romen, 2022-03-04:
Thanks for the answers. I managed to get something (very basic) more or less working.
I will keep polishing it before publishing, but I will come back with a link for feedback on improving it!

closed 2022-04-19 by TeXitoi