TeXitoi/keyberon #84: Keyberon for a kmonad-style application

jtroo, 2022-04-18:
Hello! I'm curious to know if keyberon makes sense to use as a component of a [software keyboard remapper like kmonad](https://github.com/kmonad/kmonad).

Instead of reading a key matrix for inputs and sending output over USB like a keyboard firmware would, kmonad hooks into [OS-specific keyboard hooks](https://github.com/kmonad/kmonad/blob/keycode-refactor/c_src/keyio_win.c), intercepting the raw inputs and sending processed outputs.

Not sure if an issue is the right place for this type of question, but the discussions tab seems to be disabled for this project.

Thanks!

TeXitoi, 2022-04-18:
Sure it can. It's a library, and you can use the layer object to manage the programmable thing, and you have to feed it and then read the report to send back to the OS.

jtroo, 2022-04-18:
Thanks, that's good to know!

closed 2022-04-18 by jtroo