TeXitoi/keyberon #87: Make the hard-coded 10ms polling interval configurable

riskable, 2022-05-02:
Keyberon currently has this in `hid.rs`:

```rust
impl<B: UsbBus, D: HidDevice> HidClass<'_, B, D> {
    pub fn new(device: D, alloc: &UsbBusAllocator<B>) -> HidClass<'_, B, D> {
        let max_packet_size = device.max_packet_size();
        HidClass {
            device,
            interface: alloc.interface(),
            endpoint_interrupt_in: alloc.interrupt(max_packet_size, 10), // <-- THIS
            expect_interrupt_in_complete: false,
        }
    }
...
```
Is there any way we can make that configurable?  I noticed that when I tried to implement a composite device (keyboard + mouse) I was giving the mouse a 1ms polling interval and this would result in USB enumeration errors and I couldn't figure out why.  After a deep dive I've come to find out that Keyberon's polling interval is hard-coded to 10ms.  It would be great if that was a configurable option or if there was just a 3rd argument, `interval` to pass to `new()`.

NOTE: It seems that for mouse + keyboard composite devices to work their polling intervals need to match.

TeXitoi, 2022-05-02:
Sure we can add a new_with_polling_interval or something like that.

riskable, 2022-05-02:
That sounds perfect 👍

PR #100 (2022-10-01): Add configurable HID polling rate
fixes #87
closed 2022-10-01 by TeXitoi