TeXitoi/keyberon #98: Matrix scanning pin toggle rate

camrbuss, 2022-08-21:
The toggling of the column and row pins during a matrix scan is simply done as fast as possible. With PCBs or hand wiring with high impedance combined with fast MCUs (125MHz), you can experience artifacts in the row scans where if you press `Q` on a QWERTY layout you will also get `A`. Simply slowing down the MCU clock speed typically resolves the issue, but is not a very elegant solution.

I don't know if it worth adding a delay in Keyberon, or finding another solution. Any ideas?

Current [code](https://github.com/TeXitoi/keyberon/blob/master/src/matrix.rs#L69l) that contains the issue.

Experienced the issue with the RP2040 and the [Lily58](https://www.littlekeyboards.com/products/lily58-pro-pcb-kit) and [crkbd](https://www.littlekeyboards.com/collections/corne-pcb-kits/products/corne-mx-hotswap-pcb-kit-v3) PCBs.

ChrisChrisLoLo, 2022-09-05:
Seems like you're facing what I encountered in this issue: https://github.com/TeXitoi/keyberon/issues/97.

I think you're right: the best way to go about this is to add a delay to Keyberon, and more specifically have the delay as a kind of higher order function you can pass in. I think there's a good description of what TeXitoi thinks should be done, and you're free to open a PR implementing such a thing if you get around to it. I've been too caught up with other aspects of life at the moment to give this a full look unfortunately.

TeXitoi, 2022-09-06:
Ho, right. Sorry @camrbuss didn't seen your message, but it's a duplicate of #97

closed 2022-09-06 by TeXitoi

camrbuss, 2022-09-06:
@TeXitoi  @ChrisChrisLoLo 

Thanks for identifying the duplication, I missed the first issue. I like the solution you mentioned @TeXitoi and will try to put in a PR in November.