TeXitoi/keyberon #111: Enable debugging messages through the HID interface

borisfaure, 2023-03-02:
When programming with qmk (https://docs.qmk.fm/#/), it is possible to use `printf()` to debug code when `CONSOLE_ENABLE` is set. Those messages can be read through `hid_listen` (https://www.pjrc.com/teensy/hid_listen.html).
Could the same trick be implemented in keyberon?
Do you have any pointers to do something like this when using no debugging probe?
Thanks

TeXitoi, 2023-03-02:
Keyberon is a library, so it doesn't enable any fancy thing. Without a probe, you can create a serial usb line, and log what you want in it. Every keyberon object should be Debug, so it should be quite easy.

TeXitoi, 2023-03-03:
looks like https://github.com/rgoulter/keyboard-labs/blob/master/firmware/keyberon/src/bin/minif4-36-rev2021_1-rhs.rs use a HID and a serial usb endpoint at the same time, might be an interesting base.

borisfaure, 2023-03-06:
I managed to run an example of a usb serial implementation on the chipset, as a standalone. However, I failed to have it run on the same pins as the HID implementation from Keyberon.
I tried going a different way by having 2 HIDs (one for debugging) but so far, I can only make them conflict.
Do you have any example of keyberon with 2 HIDs (one keyboard and one mouse for example)?

gzeronet, 2023-07-15:
> I managed to run an example of a usb serial implementation on the chipset, as a standalone. However, I failed to have it run on the same pins as the HID implementation from Keyberon. I tried going a [...]

Seems here is you need: https://github.com/dlkj/usbd-human-interface-device/blob/main/examples/src/bin/multi_device.rs#L83

Actually, I have let 2 HIDs (keyboard matrix and trackpoint module) work together on a stm32f401 board by sending usbd-human-interface-device keyboard & mouse reports to usb_device, and let mouse scroll button works on keyboard (but my code looks ugly).