TeXitoi/keyberon #112: Keyberon does not work with windows

tkonolige, 2023-04-03:
I can't get keyberon to work with Windows on an nrf52840. I've narrowed it down to the USB device not configuring correctly. I'm not sure why though. I've created an example here: https://github.com/tkonolige/keyberon_failed_configure that creates a keyberon usb device and a serial usb device. The keyberon device is forever in the `Default` usb device state, while serial device reaches the `Configured` state. On MacOS, both the keyberon and serial devices reach the `Configured` state. Any help or pointers for debugging would be appreciated.

TeXitoi, 2023-04-03:
The pointed repository doesn't use keyberon.

Did you tried an already validated firmware? There is some in https://github.com/TeXitoi/keyberon/blob/master/KEYBOARDS.md

The problem seems more related to nRF than keyberon. I have users that use their keyboard on Windows. Also, I don't have a nRF52840, so maybe it would be easier to ask some help on the nRF crates? And why you didn't use the hal version, that is recommended?

Another good place for help in embedded rust is the matrix channels: https://github.com/rust-embedded/awesome-embedded-rust#community-chat-rooms and especially https://matrix.to/#/#nrf-rs:matrix.org for your case.

Hope you'll find a solution, and please update the issue with the solution when you find it!

Good luck and enjoy embedded rust!

tkonolige, 2023-05-30:
The issue is that nrf-usbd will split the get device descriptor response into multiple packets unless you set a large enough maximum packet size. I've posted an issue here: https://github.com/nrf-rs/nrf-usbd/issues/14. A workaround is to manually create a usb device with a large packet size instead of using `keyberon::new_device`.

closed 2023-05-30 by tkonolige

TeXitoi, 2023-05-31:
Thanks for explaining the solution!