TeXitoi/keyberon #134: Current working uses?

LovesTha, 2024-05-19:
I'm wanting to start my own use of this library, but I'd like to reference a working use of it to help me along. But none of the keyboards in the list that I've tried have actually compiled. 

Is there one known to still work?

TeXitoi, 2024-05-19:
They was an update recently for embedded-hal 1 and usb-device 0.3 recently, it may cause your problem.

https://github.com/TeXitoi/keyberon-f4/ is up to date, cargo update from 2 month ago with embedded hal 1 and usb-device 0.3

LovesTha, 2024-05-23:
Thank you, that does build. Which is a good place for me to start tinkering and learning, thank you.

valpackett, 2024-08-22:
> none of the keyboards in the list that I've tried have actually compiled

`--locked` not being on by default in cargo strikes again, I suppose…

BTW, I recently updated my [ErgoNICE firmware](https://codeberg.org/valpackett/ergonice/src/branch/trunk/fw), which is still stuck on ehal 0.2 since I'm too lazy to update the unmaintained stm32L1 HAL to 1.0 myself. It's not the best reference for starting to learn haha, but a good one to expand into crazy territory xD