TeXitoi/keyberon #135: Shifted keys leak modifiers

senekor, 2024-08-18:
When a sequence like `&[` is rolled, the shift modifier associated with `&` is also applied to `[`, resulting in the string `&{`.

I'm looking into fixing this myself, but I wanted to document the problem anyway. Let me know if you have ideas about how to fix this in a way that aligns well with the design of the library.

senekor, 2024-08-18:
I have a working PoC with https://github.com/TeXitoi/keyberon/pull/136, but I'm happy to close that and implement anothe approach if it doesn't fit the design of the library.

TeXitoi, 2024-08-18:
I'm in vacations. I'll try to review this next week. If you don't have any sign of me in 2 weeks, don't hesitate to ping me.

I've seen this problem recently. I was thinking of.an action that will release in case of another press. You can then use it with MultipleActions.

senekor, 2024-08-18:
No worries, enjoy your vacation! I'll also try your approach in the meantime, thanks.

senekor, 2024-09-08:
@TeXitoi friendly ping :slightly_smiling_face: 

I tried to see how to implement your suggestion, but I couldn't make sense of it. As far as I can see, when a key is pressed, the other currently pressed keys do not get "notified" such that they could react by releasing themselves, right? Or did I misunderstand your suggestion?

(btw. I also opened a [PR on keyberon-f4](https://github.com/TeXitoi/keyberon-f4/pull/6) a while ago, in case you missed it)